tenstorrent/tt-exalens #329: Start using metal image as base for our docker

adjordjevic-TT, 2025-04-16:
We should start using metal image as base image for our docker

tt-vjovanovic, 2025-08-11:
Talked to Andrew Fuller, metal docker should be split into two images. One that will be base for all projects (contains build tools, etc.), and second one that is metal related. We should start using first one.

miacim, 2025-08-11:
@afuller-TT , is metal docker split happening?

afuller-TT, 2025-08-12:
It's a desirable thing to do.  But nobody is actively looking at it.

How urgent is this for you?

miacim, 2025-08-12:
@afuller-TT, the main concern is that the existing metal Docker image is about 5× larger than our current one, which could slow down tests and CI if it’s pulled often. I’m not sure yet about other dependency impacts. How complex would it be to split or slim it down — are we talking about hours, days, or a week of work?

closed 2026-01-22 by tt-vjovanovic