tenstorrent/tt-exalens #331: Add support for ubuntu-24.04

adjordjevic-TT, 2025-04-16:


blozano-tt, 2025-09-09:
I can work around it I assume, but this is problematic.

https://github.com/tenstorrent/tt-metal/actions/runs/17572041384/job/49909916425

tt-vjovanovic, 2025-09-10:
> I can work around it I assume, but this is problematic.
> 
> https://github.com/tenstorrent/tt-metal/actions/runs/17572041384/job/49909916425

Your error is coming from hardcoded value here: https://github.com/tenstorrent/tt-exalens/blob/81a1665db2f188598a38ac9b4ab77e28052452d8/setup.py#L44

One option to mitigate yourself is to do similar to `get_libjtag()`:
```python
def get_pybind():
    if os.path.exists(f"{ttexalens_home}/build/lib/ttexalens_pybind.cpython-310-x86_64-linux-gnu.so"):
        return ["ttexalens_pybind.cpython-310-x86_64-linux-gnu.so"]
    if os.path.exists(f"{ttexalens_home}/build/lib/ttexalens_pybind.cpython-312-x86_64-linux-gnu.so"):
        return ["ttexalens_pybind.cpython-312-x86_64-linux-gnu.so"]
```

onenezicTT, 2025-12-16:
Added support for this on #759

closed 2025-12-16 by onenezicTT