tenstorrent/tt-exalens #444: Add testing for DRISC

adjordjevic-TT, 2025-06-13:
In blackhole DRAM block contains risc cores (DRISC). We should investigate what can be done on them and add appropriate tests.

Since DRAM block in wormhole does not contain any risc cores tests should be skip for this case.

miacim, 2025-08-11:
@tt-vjovanovic , can you update issue with findings.

tt-vjovanovic, 2025-08-11:
There are two ways one can access DRAM noc block:
1. When GDDR is the main guy
2. When DRISC is the main guy

Case 1 is default behavior with `tt-smi -r`. This is because bit 16 is set (`1 << 15`) in register `NIU_CONFIG` (`0xFFB20100`).
If you disable that bit, case 2 will be activated.

In case 1, you can access DRISC L1 memory with address `0x20_0000_0000`. Also, you can access debug registers with `0x100_0000_0000 + register_address`.

We need to figure out how to detect which case is being used with dram and then use corresponding addresses...
Anyways, until metal decides to use Case 2, Case 1 is default and we can write test for DRISC... Also, for test, we can easily activate Case 1 and force desired behavior...