tenstorrent/tt-exalens #568: Refactor pack_unpack_regfile.py

adjordjevic-TT, 2025-08-28:
Currently in `pack_unpack_regfile.py` we have two functions for unpacking data based on whether data was read directly or through instruction injection. The only difference should be that directly read data does not need reordering bits but currently we do it as part of `unpack_data` function. We should decouple this so we can unify unpacking functions and just do reordering as needed.

onenezicTT, 2026-10-23:
@miacim @adjordjevic-TT  what's the status of this, is this still planned?

adjordjevic-TT, 2026-10-26:
> 12\. [@miacim](https://github.com/miacim) [@adjordjevic-TT](https://github.com/adjordjevic-TT)  what's the status of this, is this still planned?

yes